ruuvi/ruuvi.firmware.c #10: Cannor open the ses

Badiue, 2019-03-28:
when im try to open the ses file it say crash . im using Seeger Embedded Studio for ARM v4.12

ojousima, 2019-03-28:
Could you give more details?
 * Which tag you're on? 
 * Have you downloaded the Nordic SDK?
 * Have you updated git modules?
 * Which OS you're on?
 * Can you give the crash report?

Badiue, 2019-03-29:
im using ruuvitag
im downloaded  nRF5_SDK_15.2.0_9412b96
im alredy update git module
im using ubuntu 18.04
[image]

ojousima, 2019-03-29:
I can open the project on 4.12 on MacOSX and Windows, so this might have something to do with the linux version of SES. I don't have a Linux installation I could use to check the issue. 

Can you open the Nordic SDK examples with your SES installation?

Badiue, 2019-04-01:
yes,i can open Nordic SDK,im also have try using ruuvitag_fw and it work for this.
thank for the info,i will try using windows.

ojousima, 2019-05-22:
@Badiue Was the issue resolved with Windows?

ojousima, 2019-08-06:
Closing as stale

closed 2019-08-06 by ojousima