ruuvi/ruuvi.firmware.c #11: Where is the Gatt code?

alexe100, 2019-05-21:
Hi
I am trying to learn with this code 
I have already flash hex into ruuvi and it works but I can not find where characteristics and services were declared. 

Thanks

Alex

ojousima, 2019-05-22:
Hi,

`task_gatt` calls initialization of the GATT as defined in the ruuvi interface: https://github.com/ruuvi/ruuvi.firmware.c/blob/1a7d8b02f8bf0b776a16440395a00047721a9c6d/tasks/task_gatt.c#L94

The GATT code is then under ruuvi.drivers.c / nrf5_sdk15_platform, https://github.com/ruuvi/ruuvi.drivers.c/blob/ca917d270f327ec19eb9aec871f6a70fbf730c9e/nrf5_sdk15_platform/communication/ruuvi_nrf5_sdk15_communication_ble4_gatt.c#L509

For example the Device Information Service init function at https://github.com/ruuvi/ruuvi.drivers.c/blob/ca917d270f327ec19eb9aec871f6a70fbf730c9e/nrf5_sdk15_platform/communication/ruuvi_nrf5_sdk15_communication_ble4_gatt.c#L675 
fills the information required by DIS and then calls Nordic SDK which has the actual implementation. It should be noted that the BLE qualifications of Nordic code remain valid only if the Nordic code is used as-is.

alexe100, 2019-05-22:
It works now! 
Thanks

closed 2019-08-06 by ojousima