ruuvi/ruuvi.firmware.c #12: initialization: order of inits BLE first

DG12, 2019-08-06:
Although it seems that LEDs are easy and cannot reasonably fail, 
It would be much better to initialize BLE and start advertising earliest.
Does BLE depend on TIMER? RTC?
Consider that BLE has no external GPIO solder connections. 

GPIO, IC2 and SPI should be initialized LAST.

Is there any real-world data regarding failures? 
From production final tests?

ojousima, 2019-08-06:
BLE is initialized last because the first sent packet should have valid sensor data, and everything else has to be initialized to read the sensors. I suppose BLE could be initialized after NFC, but few tens of milliseconds doesn't make a practical difference.

BLE does depend on RTC internally, but that dependency is buried inside softdevice and not seen by application.

DG12, 2019-08-06:
BLE should send out a non-sensor data packet first.
It should contain version (including compilation date and time to permit matching with .MAP and .OUT ) and configuration information. When initialization is complete it should send out another non-sensor packet 
It's not a question of milliseconds but rather whether it can be performed at all.

Note that initialization happens every time the device resets.

ojousima, 2019-08-07:
Catching a single packet of BLE advertisement is unreliable, therefore NFC or reading the data over GATT is better suited for this kind of version information. Even if the packet is sent in every reset resetting device in hope to catch the first advertisement is a bit clumsy way to do things. 

It would be possible to match the latest git commit as a CFLAG to compiler to keep the version information in a similar manner as Jenkins pulls the git commit / tag if available, that's a good idea.

ojousima, 2019-12-17:
The automatic version information from Git Hash / commit is now implemented for ARMGCC build, closing.

closed 2019-12-17 by ojousima