ruuvi/ruuvi.firmware.c #29: static bool m_is_init; should be initialized

DG12, 2019-11-20:
in  src/tasks/communication/task_advertisement.c 

Should 
static bool m_is_init;
be 
static bool m_is_init = false;
?

ojousima, 2019-11-20:
Using uninitialized variables is a path to bugs. 
However static variables will be located in BSS-section of RAM, and startup of the firmware clears the BSS area in RAM. This is further verified by [SonarCloud](https://sonarcloud.io/code?id=ojousima_ruuvi.firmware.c&selected=ojousima_ruuvi.firmware.c%3Asrc%2Ftasks%2Fcommunication%2Ftask_advertisement.c) and [PVS Studio](https://ruuvi.github.io/ruuvi.firmware.c/fullhtml/index.html) as well as GCC analysis. None of them flag the `m_is_init` as unused variable. There's plenty of other issues though, especially in PVS MISRA-C analysis.

DG12, 2019-11-20:
You should not depend on startup of the firmware to zero variables, it's clearer anyway and it would not hurt to explicitly initialize them!

It may be advantageous at some point to disable firmware initialization that zeros BSS.
Especially with large areas of RAM for history.
A system reset, which invoke firmware initialization, might be called for BUT preserving RAM values!

For example:a WDOG expiration should NOT invoke firmware initialization if the cause can be determined and compartmentalized, but that's a whole other issue.

ojousima, 2019-11-21:
C standards define initialization along these lines: 

>If an object that has automatic storage duration is not initialized explicitly, its value is indeterminate. If an object that has static or thread storage duration is not initialized explicitly, then:
>— if it has pointer type, it is initialized to a null pointer;
>— if it has arithmetic type, it is initialized to (positive or unsigned) zero;
>— if it is an aggregate, every member is initialized (recursively) according to these rules, and any padding is initialized to zero bits;
>— if it is a union, the first named member is initialized (recursively) according to these rules, and any padding is initialized to zero bits;

The standard very clearly defines that local variables are indeterminate and static variables are zero.
Variable explicitly initialized to false will get placed into the exact same slot in application RAM and the compiler will rely on the start-up behavior to initialize the variable to zero. 

> A system reset, which invoke firmware initialization, might be called for BUT preserving RAM values!

nRF52 doesn't guarantee anything about state of the RAM after reset. It's not guaranteed to be zeroed, retained or corrupted. This is also exactly why static variables have to be zeroed at boot, alternative is to store the initial value to FLASH and then copy the initial values over to RAM from FLASH. This is also why we have to rely on MCU and compiler doing their jobs and follow the specification.

I'm not opposed to settings static variables explicitly to false value and letting compiler optimize the explicit initialization out, but unless a well-reasoned standard such as MISRA requires making it explicit I'm not going to enforce one style over other as long as the file itself is consistent.

>  For example:a WDOG expiration should NOT invoke firmware initialization if the cause can be determined and compartmentalized, but that's a whole other issue.

Watchdog must always trigger a complete reset. Watchdog is there to catch and recover from errors which aren't predicted, so by definition corrective action is to return the firmware back to known state via reboot.

DG12, 2019-11-21:
Thanks you for that information.

closed 2019-11-21 by DG12