ruuvi/ruuvi.firmware.c #33: versioning lost major.minor.update

DG12, 2019-12-17:
NFC SW now reports 7 characters from GitHub.

Automatic versioning (only) by a tool looses the intelligence behind "This is a version".

I would like to suggest that SW "version" be returned to the traditional 3 numeric field format of major.minor.update.

Perhaps an additional field can be defined containing the GitHub ID ( and a date-time too!)

DG12, 2019-12-17:
I personally prefer a compiled-on-date-time  which can be generated outside of a tool (like GitHub) and still be meaningful without access to the tool, similar to a simple include 
`uint32_t dateCompiled=0x39211512;
`
created by
`echo uint32_t dateCompiled=0x`date +%M%H%d%m\;` >dateCompiled.h # all numeric MMHHddmm
`

ojousima, 2019-12-18:
The versioning tries to find major.minor.update from GitHub, if it is not available a commit hash is used instead. 

Compilation data isn't as informative, as the developer might have checked some older version and compiled it to test if some bug was already present there

DG12, 2019-12-18:
Does "tries to find" mean that sometimes(?) there will be a  major.minor.update versions, like releases?

ojousima, 2019-12-18:
Yes, this is the exact logic:
```
# Tag on this commit
TAG := $(shell git describe --tags --exact-match)
# Commit hash from git
COMMIT := $(shell git rev-parse --short HEAD)
VERSION := $(if $(TAG),\\\"$(TAG)\\\",\\\"$(COMMIT)\\\")
```
The `VERSION` gets then passed to compiler.

DG12, 2019-12-20:
Perhaps including both in the NFC SW would be even better?

Does this make using git a requirement for development?

ojousima, 2019-12-20:
Git is not required for developing, only the makefile building the packages requires git. Any firmware version can be passed to the compiler, including the compile time.

DG12, 2019-12-20:
Version and Build as Apple Mac OS
[image]

ojousima, 2019-12-23:
Does this issue need some further discussion or work?

DG12, 2019-12-24:
How do you feel about including both version and build to NFC SW ?

ojousima, 2019-12-25:
Eventually the official builds are going to be pushed straight from Jenkins to GitHub Releases, so adding the build information isn't necessary from Ruuvi's point of view. Anyone is of course free to add additional information to their builds as they see fit.

closed 2020-01-02 by DG12