ruuvi/ruuvi.firmware.c #65: ruuvi_interface_lis2dh12.c:175 FATAL: SELFTEST

rinat-enikeev, 2020-01-21:
I've faced the problem with 2 of my RuuviTags - in connected state they were restarting with the 

> ruuvi_interface_lis2dh12.c:175 FATAL: SELFTEST

error. Re-programming didn't help. Next morning the issue was "fixed". Without any action from my side. 

The tag advertised ok, but didn't heartbeat.

ojousima, 2020-01-21:
The LIS2DH12 lacks a proper soft reset, I've seen some bus related to LIS2DH12 survive reboots and reprogramming tags. I'll start by checking that a failed selftest cleans up the state of LIS2DH12.

DG12, 2020-01-21:
My 06DC tag after reset reports:
Program start
Version: v3.28.13
'mcu_tests':{
'library':{
'peak2peak':{
'valid_data':'pass',
'nan_data':'pass',
'overflow_data':'pass',
'input_validation':'pass',
},
'rms':{
'valid_data':'pass',
'nan_data':'pass',
'overflow_data':'pass',
'input_validation':'pass',
},
'variance':{
'valid_data':'pass',
'nan_data':'pass',
'overflow_data':'pass',
'input_validation':'pass',
},
'total_tests':'15',
'passed_tests':'15'
}
}
FPU Error 
Running extended self-tests, this might take a while
test_sensor.c:33 WARNING: NOT_FOUND
test_environmental.c:60 WARNING: SELFTEST
Tests ran: 85, passed: 84
Radio is already in use by other module - beware coexistence issues
<info> app: Setting vector table to bootloader: 0x00075000
<info> app: Setting vector table to main app: 0x00026000
Stored data
Stored data

ojousima, 2020-07-14:
Fixed, marking as pending until merged to master

commit 9e530732 (2020-07-14): Fix #65, Fix #83, Fix #86
closed 2020-09-09 by ojousima