ruuvi/ruuvi.firmware.c #83: LIS2DH12: Initialization failure on first boot

ojousima, 2020-07-09:
Sometimes on first boot LIS2DH12 returns a test failure.

opsationby, 2020-07-13:
@ojousima Under what conditions is the problem reproducing? Now I can reproduce the accelerometer error for moving the sensor during the self-test, but  physically move the device during the test must be excluded...

ojousima, 2020-07-13:
@opsationby It might be about physically moving the device, I've also seen initialization fail at power on. A pretty reliable way to reproduce the error is to power off the devkit and re-program the devkit.

opsationby, 2020-07-14:
I think https://github.com/ruuvi/ruuvi.drivers.c/pull/102 may fix this problem

ojousima, 2020-07-14:
@opsationby Yes, thanks. I'll mark this as pending merge to master.

commit 9e530732 (2020-07-14): Fix #65, Fix #83, Fix #86
closed 2020-09-09 by ojousima