ruuvi/ruuvi.firmware.c #86: Battery measurement works only on first sample(s)

ojousima, 2020-07-09:
Battery measurement is synchronized to radio, ADC is configured for sampling on before_radio_activity event and actual sample is taken on after_radio_activity event. The sampled data is stored in `rt_adc` and read later in `app_heartbeat` when sending data as BLE advertisement. 

The ADC was uninitialized in 3.28 firmwares after sample to reduce power consumption. Now either uninitialization or reinitialization errors and the data won't be updated. 
```
app_sensor.c:236 WARNING: INVALID_STATE
app_sensor.c:231 WARNING: BUSY
app_sensor.c:236 WARNING: INVALID_STATE
app_sensor.c:231 WARNING: BUSY
```

Relevant functions are in [app_sensor](https://github.com/ruuvi/ruuvi.firmware.c/blob/b34535516dca8d16b421d45d9acdc8f45b1c5a28/src/app_sensor.c#L215), [app_heartbeat](https://github.com/ruuvi/ruuvi.firmware.c/blob/b34535516dca8d16b421d45d9acdc8f45b1c5a28/src/app_heartbeat.c#L47) and [ruuvi_task_adc](https://github.com/ruuvi/ruuvi.drivers.c/blob/bd09d011ba3d08a05f29a84e805ed6dc3d6e2369/src/tasks/ruuvi_task_adc.c#L250). 

The sampling after radio transmission should be fixed. Ruuvi Station app can be used to check that ADC data is getting updated.

DG12, 2020-07-09:
Can you please revisit "... initialize variables to compile time values from default VALUES and permit subsequent configuration (without compile) "

as per https://github.com/ruuvi/ruuvitag_fw/issues/145 and perhaps https://github.com/ruuvi/ruuvitag_fw/issues/137
Thank you

ojousima, 2020-07-10:
e.g. sensor configuration is already stored on flash, and things such as data format and radio power / interval are on roadmap.

opsationby, 2020-07-14:
@ojousima I think ruuvi/ruuvi.drivers.c#105 may fix this problem

ojousima, 2020-07-14:
@opsationby Yes, this issue is fixed. I'll leave this issue open until update-modules is merged to master, thanks.

commit 9e530732 (2020-07-14): Fix #65, Fix #83, Fix #86
closed 2020-09-09 by ojousima