ruuvi/ruuvi.firmware.c #89: Feature: Radio configuration over GATT

ojousima, 2020-07-10:
e.g. data format, transmission interval, PHYs used, channels used.

DG12, 2020-07-10:
It seems to me that NFC would be less susceptible to vandalism due to the required proximity and keeping the communications away from BLE would also simplify implementation on both the ruuvi tag and the application to do the setting.

ojousima, 2020-07-10:
True, there should be some kind of security to unlock DFU updates and configuration.

DG12, 2020-07-10:
If you're near enough to do NFC then you are near enough to grab the tag and do whatever you want. Simple security to avoid accidentally alterations seems al that is necessary.

theBASTI0N, 2021-03-03:
You should be able to add passkey to tag just like eddystone tags can be configured with. That would prevent users from changing settings and flashing the device if passkey is not knwon.

ojousima, 2021-03-03:
Currently NFC scan is required to enable BLE bootloader, I have been thinking about using the unique ID of tag as a passkey. The ID will be exposed to anyone with NFC reader, but I think it would be secure enough as attackers need physical access to tag to configure it.