ruuvi/ruuvi.firmware.c #95: Feature: Temperature,.. (env sensors )logging

jplajpla23, 2020-07-16:
Needeed:
Env sensors logging in a circular array every x minutes (configurable) and when mobile app connecto to tag get all logs.
if no smartphone connect and circular array go to end they back to init and replace the older by new.

ojousima, 2020-07-16:
This actually is a part of released 3.28.x versions, you can find the release at https://github.com/ruuvi/ruuvi.firmware.c/releases/tag/v3.28.13. ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_ files are probably best fit for you, they store 3.5 days (1024 samples at 5 minutes) of data. Details of communication over GATT are at https://docs.ruuvi.com/communication/bluetooth-connection and https://docs.ruuvi.com/communication/bluetooth-connection/nordic-uart-service-nus/log-read . 

Example on how to read the logs with iOS is at https://github.com/rinat-enikeev/BTKit . 

The data formats might still change, the current format requires 3 UART transmissions for one sample which is not the most efficient way of transferring logs. Contact me at otso@ruuvi.com or @ojousima at #ruuvitag in slack.ruuvi.com if you're starting application development which uses the logged data so we can discuss the timelines and data formats in detail.

DG12, 2020-07-16:
Is there any planed or in progress in 3.29 to increase the number of samples stored?

ojousima, 2020-07-17:
>Is there any planed or in progress in 3.29 to increase the number of samples stored?

Yes, samples will be compressed and stored to flash

jplajpla23, 2020-07-17:
@ojousima 
what is the diferences between the all versions in the link.
https://github.com/ruuvi/ruuvi.firmware.c/releases/tag/v3.28.13

is possible configure the read interval (5m minutes configura more or less
)

ojousima, 2020-07-17:
@jplajpla23 the versions are for different hardware boards and configurations. 
_debug versions are for developers, they send and log data at a fast rate and print debug messages to Segger RTT. 
_default versions are a good fit for most users
_longlife versions broadcast at a rare interval for long battery life, but it's very slow to connect to them.
_longmem versions log at 12 hours interval. 

_full.hex versions can be flashed with a wired connection,
_app.hex is only the application without bootloader or softdevice,
_sdk_dfu versions can update 1.x and 2.x FW RuuviTags bought from Ruuvi Shop to use newer softdevice and bootloader
_dfu versions can update tags on 3.x FW. 

Logging interval is currently hardcoded, we might make it configurable over GATT in the future.

jplajpla23, 2020-07-17:
@ojousima 
OK, the 5 minutes interval version what is the expectable batery life?

ojousima, 2020-07-17:
@jplajpla23 Depends on hardware, on RuuviTag+ with BME280 and LIS2DH12 2-3 years while not connected and 6 months while connected with GATT. As the firmware is in alpha you should verify the functionality before installing it on a lot of units or wait for to us to make a version which we have verified.

jplajpla23, 2020-07-17:
@ojousima  Ok

ojousima, 2020-07-17:
>> you have some simple example to get all logs on connect ?

> Example on how to read the logs with iOS is at https://github.com/rinat-enikeev/BTKit .

> https://docs.ruuvi.com/communication/bluetooth-connection/nordic-uart-service-nus/log-read

I also have made this Web Bluetooth page to read logs: https://ojousima.github.io/ojousima.ruuviwebble.ts/
Source of Web Bluetooth implementation is at https://github.com/ojousima/ojousima.ruuviwebble.ts but it's only for my own experimentation and not recommended for anything.

jplajpla23, 2020-07-17:
@ojousima OK thanks, if the beacon fill the logs and smrtphone has not in the area to downalod the logs the beacon stop the readings until download the data or beacon start removing from the begin one position by each on reading on end?

ojousima, 2020-07-17:
@jplajpla23 The logs are currently stored to a ringbuffer, if there is no room for a new sample oldest sample is overwritten.

jplajpla23, 2020-07-27:
@ojousima Hi again, the firmware with Log maintains transmission to the broadcast of values in real time of sensors

ojousima, 2020-07-27:
Yes, real-time data is broadcasted unless there is a connected device. In future the realtime data is broadcasted during connections too.

jplajpla23, 2020-08-13:
@ojousima 
Hi again i make some updates in app and now I will use too the broadcast values, your firmware use waht Ruuvi format for decode the values.

ojousima, 2020-08-13:
Hello @jplajpla23, You should support at least data formats 3 and 5.

jplajpla23, 2020-08-21:
Hi again @ojousima, is possible after read all logs clean the buffer/array of logs to remove and not download again?

jplajpla23, 2020-08-26:
@ojousima, Is possible do a version with Log every minute?
have some problem make the log evvery minute instead 5 minutes, 
problemas like batery duration?

ojousima, 2020-08-26:
@jplajpla23 there's no problem in logging once per minute, especially on 3.28.x firmwares which have the log in RAM. In 3.29.x (still under development) data is logged to flash and the flash erase/writes will consume battery a bit faster, but I don't have numbers on the exact consumption yet.

jplajpla23, 2020-09-02:
@ojousima, Hi can help me explain me where can i update in firmware code  to read all minutes instead of each 5 minutes
Thanks

ojousima, 2020-09-02:
> @ojousima, Hi can help me explain me where can i update in firmware code to read all minutes instead of each 5 minutes

In current master: 

https://github.com/ruuvi/ruuvi.firmware.c/blob/4a1b1057fe5f63b1e91076023bd7e8910bb84241/src/application_config/application_mode_default.h#L227

Or if you're willing to preview the next version on branch update-modules,
https://github.com/ruuvi/ruuvi.firmware.c/blob/afeda5ba7d2384bd26ec4a5148c561c24fbb6060/src/application_config/app_config.h#L246

jplajpla23, 2020-09-02:
@ojousima what is the diference between ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_dfu_app.zip
and kaarle_armgcc_ruuvifw_default_v3.28.13_dfu_app.zip
 version

ojousima, 2020-09-02:
> what is the diference between ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_dfu_app.zip
and kaarle_armgcc_ruuvifw_default_v3.28.13_dfu_app.zip

Hardware version, we have some PCBs which aren't available for sale at least yet. RuuviTag binaries are the ones you want to use.

jplajpla23, 2020-09-02:
@ojousima  do you knwo some mobila android app to connect and make some communication to test the log?

ojousima, 2020-09-02:
> do you knwo some mobila android app to connect and make some communication to test the log?

I use nRF Connect

jplajpla23, 2020-09-02:
>I use that to update the firmware by dfu,
but if after update the buttton to connect not appear, only when in boot mode

ojousima, 2020-09-02:
Have you verified that you're on 3.28.13 FW version? e.g. scan the NFC of tag?

jplajpla23, 2020-09-02:
I check and and FW is not te 3.28.13, now I cloe the repo to update the firmaware to 1 minute, exists some manual to build the firmware to after I upload by DFU?

ojousima, 2020-09-02:
@jplajpla23 please join Ruuvi Slack (slack.ruuvi.com) or Ruuvi Forums (f.ruuvi.com) for longer discussion, this isn't really related to the original subject of logging feature anymore. In short, you should be able to update the firmware via DFU using these instructions https://lab.ruuvi.com/dfu/. Use https://github.com/ruuvi/ruuvi.firmware.c/releases/download/v3.28.13/ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_sdk12.3_to_15.3_dfu.zip if you're on 2.x Ruuvi Firmware, https://github.com/ruuvi/ruuvi.firmware.c/releases/download/v3.28.13/ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_dfu_app.zip if you're on another 3.x firmware and https://github.com/ruuvi/ruuvi.firmware.c/releases/download/v3.28.13/ruuvitag_b_armgcc_ruuvifw_default_v3.28.13_full.hex if you're using a wired programmer.

closed 2020-09-02 by jplajpla23

jplajpla23, 2020-09-03:
@ojousima Hi ojousima,
after update the firmware I alread can comunicate,
after send the example (https://docs.ruuvi.com/communication/bluetooth-connection/nordic-uart-service-nus/log-read#data-endpoints)

Writing in caracteristic the hex value 30 30 11 5D 67 40 ED 5D 57 FE AD (example in link)
I receive a lot of notifications the values like 05-17-C3-3A-9B-C7-D8-FF-E4-FF-FC-03-F8-B6-D6-13-07-26 
and not like the docs and log example in link with values like 3A-32-10-5D-C1-4A-A6-00-01-84-40 or based on format of link

0x3A 30 10
5D57FEAD 000000098D

ojousima, 2020-09-03:
> I receive a lot of notifications the values like 05-17-C3-3A-9B-C7-D8-FF-E4-FF-FC-03-F8-B6-D6-13-07-26

These are current sensor data, you'll need to wait a while for the logged data to accumulate before reading it out.