ruuvi/ruuvi.firmware.c #99: Store measurements in integer hundredths not float [question]

DG12, 2020-08-24:
in /app_log.h 
Would it be better to keep measurements in app_log_element_t in integer hundredths rather than float?
I wonder if there are sever compression considerations especially dealing with successive readings that are not represented well in hexadecimal fractions which are actually nearly the same.

ojousima, 2020-08-25:
The goal is to add compression later on, current compression algorithm uses too much ram to decompress and compress data at the same time. This leads to need to flush current block to flash before starting decompression, which leads to oldest data being overwritten and leaves an under utilized block in flash.

DG12, 2020-08-27:
Before adding compression, storing measurements in fixed point (just as in the transmission ) would require each element in the structure to take much less storage. Basically float point is best used when the range of possible values varies very greatly (i.e. orders of magnitude ). When the range is small floating point wastes 1 of 4 bytes for an exponent.

ojousima, 2020-08-28:
Yes, I'd consider reducing the data type size to be a part of compression.

ojousima, 2020-09-18:
Marked this as proposed enchantment. A good moment to work on this would be milestone 3.30.0 when acceleration logging is worked on. ref #24

DG12, 2021-01-07:
Further reduction by storing timestamp in tenth of seconds and only using a 3 byte field allows range of up to 71 days. At 1 entry per second far this far exceeds the flash capacity. That is, if the RuuviTag is logging a sample every second and the log is circular and  the RuuviTag  must be recovered within 71 days.

Using the advertising format this would use the same number of bytes as float t,h,p and be able to  include t,h,p,x,y,z,move.

ojousima, 2021-02-10:
Closing this as it will not be worked on near future.

closed 2021-02-10 by ojousima