ruuvi/ruuvi.firmware.c #100: Feature: Allow configuring settings via GATT connection only for a certain time after boot

laurijamsa, 2020-08-27:
Make it not possible to configure settings wirelessly after a hard-coded time slot has closed.

DG12, 2020-08-27:
Is the intent of this to prevent malicious modification?

Why had coded time limit rather than a configurable limit?

ojousima, 2020-08-28:
> Is the intent of this to prevent malicious modification?

yes

> Why had coded time limit rather than a configurable limit?

It can be configurable as in a constant in app_config.h, but configuring it via GATT would add unnecessary complexity for 99% of use-cases.

ojousima, 2020-09-11:
Problem with enabling configuration after boot is that tags start to reboot every now and then once battery is low, and attacker could intentionally drain the battery by repeatedly reading logs. 

Also if attacker finds a way to crash the tag with some BLE command after reboot it would let them gain full access. There should be a physical contact to device to allow configuration. 

Linked issues #105 #107 #106

ojousima, 2020-09-16:
The configuration is now possible after pressing button and NFC reads. As RuuviTags have at least a button and opening configuration at boot would be dangerous when battery is drained, let's close this issue and revisit it if there's device without button or NFC option.

closed 2020-09-16 by ojousima