ruuvi/ruuvi.firmware.c #107: v3.29.0-RC2: DFU unlocking through NFC is not implemented.

ojousima, 2020-09-10:


DG12, 2020-09-15:
What about requiring a password. It could be stored in UICR, defaulted to FFs. 
Allow GATT (or NFC) changing password and other values (as well as enabling DFU) if password is currently default or if matches previously set.
This would allow users with security concerns to use a password of their choosing and other users ease of configuration.
Improperly passworded tags could be recovered with physical access and  nrfjprog --eraseuicr

ojousima, 2020-09-15:
> What about requiring a password. It could be stored in UICR, defaulted to FFs.

We have been thinking about using the DeviceID as a default password, as it's strong enough and unique to each device. As an added bonus there's already mechanism for reading it out via NFC and GATT, ref #117 and #109. 

> Improperly passworded tags could be recovered with physical access and nrfjprog --eraseuicr 

This would be really bad user experience for those without access to devkit, we're going for a factory reset + enter bootloader by holding down button for a while. That should allow users without devkits to recover their devices.

commit 9843f176 (2020-09-15): ISSUE RF-35 fix #107 (#118)

* Add NFC disconnected callback, refactor configuration enable/disable

* Refactor bidirectional communications out of builds which don't use them

* Enable configuration on next connection after NFC disconnect

* Fix code smells
closed 2020-09-15 by ojousima