ruuvi/ruuvi.firmware.c #156: 3.29.0: Integration test report isn't entirely valid JSON

ojousima, 2020-10-14:
Boards which are missing sensors report "WARNING: NULL", some colons are missing, some colons are before closing double quotes

ojousima, 2021-01-25:
A sample JSON:
```
{
"firmware":"Ruuvi FW ",
"board":"RuuviTag B",
"ruuvi.boards.c":"3.5.0",
"ruuvi.drivers.c":"3.5.0",
"ruuvi.libraries.c":"3.0.0",
"libraries": {
"peak2peak":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC Error",
"overflow_data":"pass",
"input_validation":"pass"
},
"rms":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC ErrorFPU OFC Error",
"overflow_data":"pass",
"input_validation":"pass"
},
"variance":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC ErrorFPU OFC Error",
"overflow_data":"pass",
"input_validation":"pass"
},
"ringbuffer":{
"put_get":"pass",
"overflow":"pass",
"overdrain":"pass"
},
"total_tests":"15",
"passed_tests":"15"
},
"drivers": {
"flash":{
"init":"pass",
"uninit":"pass",
"store":"pass",
"load":"pass",
"free":"pass",
"gc":"pass"
},
"power":{
"dcdc_internal":"pass"
},
"timer":{
"init":"pass",
"create":"pass",
"single_shot":"pass",
"repeat":"pass"
},
"scheduler":{
"init":"pass",
"execute":"pass",
"put":"pass"
},
"sensors": {
"SHTCX": {
"init":"pass",
"modes":"pass",
"configuration":"pass",
"interrupts":"skipped"
"data":{
"timestamp_ms": "88560",
"humidity_rh": "34.78",
"temperature_c": "29.27",
}
},
"ruuvi_nrf5_sdk15_log.c:68 WARNING: NULL
": {
"init":"skip",
},
"BME280": {
"init":"pass",
"modes":"pass",
"configuration":"pass",
"interrupts":"skipped"
"data":{
"timestamp_ms": "89089",
"humidity_rh": "35.69",
"pressure_pa": "99113.89",
"temperature_c": "29.16",
}
},
"LIS2DH12": {
"init":"pass",
"modes":"pass",
"configuration":"pass",
"interrupts":"fail"
"data":{
"timestamp_ms": "91560",
"acceleration_x_g": "0.07",
"acceleration_y_g": "0.03",
"acceleration_z_g": "1.00",
"temperature_c": "34.50",
}
}
},
"radio":{
"init":"pass,"
"address_set_get":"pass"
},
"ble_adv_1_mbit":{
"init":"pass",
"interval":"pass",
"extended":"pass",
"power":"pass",
"scan":"pass"
},
"ble_adv_2_mbit":{
"init":"pass",
"interval":"pass",
"extended":"pass",
"power":"pass",
"scan":"pass"
},
"ble_gatt_1_mbit":{
"init":"pass",
"tx":"timeout"
"throughput":"0.000 bps"
},
"ble_gatt_2_mbit":{
"init":"pass",
"tx":"timeout"
"throughput":"0.000 bps"
},
"nfc":{
"init":"pass",
"tx_rx":"timeout"
},
}
}
```
Issues can be found with e.g. https://jsonformatter.curiousconcept.com/ , paste JSON and press "process"

DG12, 2021-01-25:
Is there a program you want to feed the JSON to?

ojousima, 2021-01-25:
Not yet, longer term I'd like to have the tests run and checked automatically to reduce repetitive manual work on releases

DG12, 2021-01-25:
Suggestion regarding [https://docs.ruuvi.com/ruuvi-firmware/3.x/releases](https://docs.ruuvi.com/ruuvi-firmware/3.x/releases)

I would suggest

1a) "battery death" test , remove and insert good battery
1b)  "nearly battery dead" test with known feeble battery.
These should be program directed, readings compared to expected standard  but start by hand .
2a) environmental heat test. Leave in oven at 80C for 5(?) minutes verify T & H sensor reading
2b) environmental cold test. move to freezer -xxxC for 5(?) minutes  verify T & H sensor reading at 1,2,3,4,5 minute and compare to standard document.
3) with Ruuvi flat rotate clockwise for 1/4 hold for count of 3 (or 3 seconds more or less), rotate 1/4 turn so inverted and hold, rotate 1/4 turn... return to original flat. 
4) With ruuvi on edge with indicator dot ( well if there isn't one on the enclosure, put one on with a marker) at table surface. Roll through 2 revolutions pausing at each 90 degree. compare with standard

3) "Missing sensors are allowed." should be more specific.

Maybe you could put this test by test in git hub?

ojousima, 2021-01-26:
> 1a) "battery death" test , remove and insert good battery
1b) "nearly battery dead" test with known feeble battery.
These should be program directed, readings compared to expected standard but start by hand .

What exactly this would test?

> 2a) environmental heat test. Leave in oven at 80C for 5(?) minutes verify T & H sensor reading
2b) environmental cold test. move to freezer -xxxC for 5(?) minutes verify T & H sensor reading at 1,2,3,4,5 minute and compare to standard document.

These are related more to hardware than software testing

> 3a)with Ruuvi flat rotate clockwise for 1/4 hold for count of 3 (or 3 seconds more or less), rotate 1/4 turn so inverted and hold, rotate 1/4 turn... return to original flat.
3b) With ruuvi on edge with indicator dot ( well if there isn't one on the enclosure, put one on with a marker) at table surface. Roll through 2 revolutions pausing at each 90 degree. compare with standard
3c) record movement counter

The movement counter is only indicative of "movement has happened", it doesn't have any applicable standard for how much it should increment on some specific movement

> Maybe you could put this test by test in git hub?

It's in docs.ruuvi.com which gets stored to https://github.com/ruuvi/docs

DG12, 2021-01-26:
1a and 1b simulate real world events which the software needs to be able to recover from. We know that the battery responds to temperature and that the software needs to be able to recover if the battery temporarily drops below operating voltage and returns.

Although this feels like a hardware test, It is a final validation that the software (actually the entire system) is properly reporting the things it is designed to.
 
These test should be done, maybe just once, before any release is finalized.

DG12, 2021-01-26:
I know it's a little late now but please consider the difficulty of outputting JSON.
I would like you to consider a simple name=value (no quotes necessary) format like that used by html submit.
JSON seems excessively complicated, general and very awkward to code with it's many quotes and brackets.
Hierarchy can easily be included with parenthesis only where needed.
This would also reduce the output by nearly half in both time and space and make the output much less noisy.
Consider that as additional or alternate devices are added more if this will need to be coded.

ojousima, 2021-03-19:
@niksiboxi Please check if this can be closed now

niksiboxi, 2021-03-19:
Tested: JSON packet prints without issues.

closed 2021-03-19 by niksiboxi