ruuvi/ruuvi.firmware.c #163: 3.29.0-RC10: Poor GATT connection might lead to watchdog reboot

ojousima, 2020-10-15:
On a poor connection, it takes over 120 seconds to transfer logs. This will lead to watchdog triggering and reboot, with associated loss of logged data. Applications should always disconnect at 60 seconds tx is ongoing, and firmware could kick the connection at 90 seconds or so.

ojousima, 2020-11-05:
For now, this will be the responsibility of the reader. If log transfer takes over 60 seconds, reader should disconnect.

ojousima, 2021-03-19:
Fixed, tag kicks connection before watchdog timer triggers.

closed 2021-03-19 by ojousima

DG12, 2021-03-19:
Why not add a occasional dog feed ?

ojousima, 2021-03-19:
If there is a bug which causes log processing to loop forever, watchdog needs to trigger.

DG12, 2021-03-19:
Perhaps a better solution to this particular known case (which is not really a bug) is to incorporate a timeout in the GATT transmission. The reader who doesn't even know that there is a problem until it's too late.  The reader can move to a more "friendly" (closer) location and try again or at a different time when there is less interference.

 The wdog reset will cause all historical data to be erased. 

Relying on the reader, which might not be a Ruuvi app but written by the user, doesn't sound like a good idea.

ojousima, 2021-03-19:
> Perhaps a better solution to this particular known case (which is not really a bug) is to incorporate a timeout in the GATT transmission. The reader who doesn't even know that there is a problem [...]
The wdog reset will cause all historical data to be erased.
Relying on the reader, which might not be a Ruuvi app but written by the user, doesn't sound like a good idea.

> Fixed, tag kicks connection before watchdog timer triggers.

Already done