ruuvi/ruuvi.firmware.c #168: logging: Significant improvement in time period reported by internal duplicate reduction - improvment

DG12, 2020-11-05:
The actual environment (not the readings produced by the limited accuracy of the sensors ) in many cases, varies very slowly in relation to the period between log samples.  It would be rather simple to use an illegal value of one field to indicate that the previous values are repeated multiple times. This has the potential of greatly extending the length of time the in memory log covers.

For example use a value for temperature of 86C (well outside the operating range of the tag but still within the range of the field allocated) and then have the next field (in the current implementation humidity) contain 20. At the current period of storing log entries every 5 seconds that would indicate that the environment had not changed for 100 seconds. This algorithm takes advantage of an understanding  of the application which provides compression far beyond any generalized compression scheme.

This could be expanded at during syncing which would avoid any changes to the log receiver application.

This is In addition to the suggestion of storing log entries as fixed point in the same format as they are transmitted in the heart beat packets discussed in issue#99

ojousima, 2020-11-05:
One of the core requirements of the logging is to guarantee the number of samples / logged time. Any variable-length output would mean that we cannot guarantee a specific logged time, and would have to say e.g. "History has 10 000+ values". Compressing data was piloted in #73, but overall handling the compression + decompression caused more problems than benefits. 

The core issue is "inefficiently stored data", but as the internal buffer already has space for over one week we're going to apply [YAGNI](https://en.wikipedia.org/wiki/You_aren%27t_gonna_need_it) here and revisit it if an use case where one week of logged data isn't enough comes up.

DG12, 2020-11-05:
Please notice my sentence:

This could be expanded at during syncing which would avoid any changes to the log receiver application.

This would allow space for additional sensor history ( especially x,y,z and movements0

ojousima, 2020-11-06:
> Please notice my sentence:
> This could be expanded at during syncing which would avoid any changes to the log receiver application.

Doesn't solve 
>> Any variable-length output would mean that we cannot guarantee a specific logged time, and would have to say e.g. "History has 10 000+ values". 

> This would allow space for additional sensor history ( especially x,y,z and movements0

Yes, we'll revisit data compression once the need arises

DG12, 2020-12-31:
Closed better mechanism to store in advertise format see issue #99

closed 2020-12-31 by DG12