ruuvi/ruuvi.firmware.c #173: heartbeat does not preserve intermediate err_code - nit

DG12, 2020-11-21:
in src/app_heartbeat.c function heartbeat 

`err_code = ....`

causes previous err_code values to be lost

Should Be 
`err_code |=`

Just saying.

ojousima, 2020-11-23:
Do you mean function https://github.com/ruuvi/ruuvi.firmware.c/blob/ba564b6e19fc5bd1598852a64b0ef4da19f14060/src/app_heartbeat.c#L94? 

The err_code is intentionally discarded after checking if watchdog can be fed. For example sending data via gatt fails always when there is no connection and there's nothing that needs to be done for it.

DG12, 2020-11-23:
L111 send_adv "should always be successful" and set headrtbeaet_ok=true;
L124 if rt_gatt_send_asynchronous  sets error_code heartbeat_ok remains TRUE;
L133 rt_nfc_send will lose error from rt_gatt_send_asynchronous heartbeat_ok REMAINS TRUE!
L143 app_log_process loses any previous error

Can any of these actually fail anyway?

ojousima, 2020-11-23:
This is intentional, Known failure states are:

* L111: Advertisement is sent while connection is being established, before NUS registration is complete. Application tries to send connectable advertisement which gets rejected by drivers. 
* L124: NUS connection is not established.
* L133: nRF52811, has no NFC hardware. 

L143 is intentional, we only need to check if tag can communicate. It could be argued that NFC success should not set the ok flag, though.

DG12, 2020-11-23:
Perhaps some  similar comments in the source might be helpful ?
 NUS??

ojousima, 2020-11-24:
NUS stands for Nordic UART Service. 
A proper place for comment would be at https://github.com/ruuvi/ruuvi.firmware.c/blob/ba564b6e19fc5bd1598852a64b0ef4da19f14060/src/app_heartbeat.h#L35, and it really should be documented that heartbeats must be started to feed watchdog and under which conditions watchdog gets fed. Adding comment inside implementation is probably not necessary.

ojousima, 2020-11-24:
Updated the title to reflect the documentation.

PR #247 (2021-04-12): Add watchdog "Feed" description
Resolve #173
closed 2021-04-12 by niksiboxi