ruuvi/ruuvi.firmware.c #176: 3.29.1: Old configuration remains in flash

ojousima, 2020-12-10:
Kind of feature/bug, a tag which has been running debug variant remembers debug settings and continues with those and vice versa.

ojousima, 2021-03-18:
@niksiboxi Please verify if this has been fixed.

Steps: 
1) Flash 3.29.3_test_full.hex to RuuviTag.
2) Update latest 3.30-RC release version via DFU. 
3) Press and hold "B" to erase old settings. Hold at least 10 seconds and wait for the tag to reboot. If tag enters bootloader, just reset it. 
4) Check that logged data interval is 5 minutes (Ruuvi Station) and broadcast interval is 1.285 s (nRF Connect). 

Close this issue if it's been fixed, if not, let's get it fixed :)

niksiboxi, 2021-03-18:
@ojousima Can't do DFU:
- Flashed with `ruuvitag_b_armgcc_ruuvifw_test_v3.29.3_full.hex`.
- Downloaded 3.30-RC zip file on Android phone. 
- Set RuuviTag to bootloader mode.
- Tried to do DFU, but it doesn't work: the nRF Connect starts to upload, but cancels immediately.

niksiboxi, 2021-03-19:
@ojousima Tested with `ruuvitag_b_armgcc_ruuvifw_default_v3.30.0-RC5_dfu_app.zip`:
- Broadcast interval is 1.285'ish (nRF Connect)
- Logged data interval is every second on Ruuvi Station.

niksiboxi, 2021-03-19:
Verified with Ruuvi Station app: data is logged every 5 min.

closed 2021-03-19 by niksiboxi