ruuvi/ruuvi.firmware.c #182: 3.29.3 movement counter updated by 2 on every other movement - defect; priority:L

DG12, 2020-12-18:
Running test version with j-link and JLinkRTTClient
Moving RuuviTag slightly. Log report movement but movement counter is not incremented.
Move slightly again, log reports movement and movement counter is incremented by 2 !

ojousima, 2020-12-18:
That's curious, has this been checked against measurement sequence counter of advertisements? Thanks for the report, I'll check the encoding and counter logic anyway and try to repeat

DG12, 2020-12-18:
This does not occur in 2.5.9

DG12, 2020-12-18:
Consequence of this is that a single movement  is not reported until another movemet occurs which might be a long time later.

ojousima, 2020-12-18:
I can repeat on a devkit

ojousima, 2020-12-18:
Fixed in #185, root cause was that tests didn't verify params going to encoding. Fix will be included in next release.

commit 6fa363db (2020-12-18): Fix #182 (#185) - Movement counter increments only to even values
closed 2020-12-18 by ojousima