ruuvi/ruuvi.firmware.c #183: 3.29.3  some log messages during initial HW test appear to be errors - question: priority:L

DG12, 2020-12-18:
Do the **bold** parts here indicate a problem ( with the hardware) ?
System seems to be running fine.
"ruuvi.boards.c":"3.4.0",
"ruuvi.drivers.c":"3.4.5",
"ruuvi.libraries.c":"3.0.0",
"libraries": {
"peak2peak":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC **Error**",
"overflow_data":"pass",
"input_validation":"pass"
},
"rms":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC ErrorFPU OFC **Error"**,
"overflow_data":"pass",
"input_validation":"pass"
},
"variance":{
"valid_data":"pass",
"nan_data":"pass",
"overflow_fpu_errors":"FPU OFC ErrorFPU OFC **Error**",
"overflow_data":"pass",
"input_validation":"pass"
...
"LIS2DH12": {
"init":"pass",
"modes":"pass",
"configuration":"pass",
"interrupts":"**fail**"

"ble_gatt_1_mbit":{
"init":"pass",
"tx":"**timeout**"
"throughput":"**0.000 bp**s"
},
"ble_gatt_2_mbit":{
"init":"pass",
"tx":"**timeout**"
"throughput":"**0.000 bp**s"
},
"nfc":{
"init":"pass",
"tx_rx":"**timeout**"
},

ojousima, 2020-12-18:
`"overflow_fpu_errors":"FPU OFC ErrorFPU OFC Error",` These are intentional, the calculation overflows float's max, have divide by zero etc. The test verifies that nothing crashes and errors get reported to application.

`"interrupts":"fail"` This is related to level and FIFO interrupt tests not being separated, it's a problem in test itself. Related issues https://github.com/ruuvi/ruuvi.drivers.c/issues/167 and https://github.com/ruuvi/ruuvi.drivers.c/issues/107

`"tx":"timeout"` These tests require nRF52 DK programmed with the hex at system_tests folder to connect for measuring the GATT throughput. 

NFC requires an external reader copying the contents of NFC, i.e. read + write back the read data.

DG12, 2021-01-07:
BLE & NFC timeout:  MAC:....1A:96. Ruuvi FW v3.29.3+debug. 
PLEASE reduce the ble_gatt_1_mbit Tx test time. RuuviTag got hung and I waited a long time to complete test!!

Second run(today)No timeout occured

First run (today):
"firmware":"Ruuvi FW ",
"board":"RuuviTag B",
"ruuvi.boards.c":"3.4.0",
"ruuvi.drivers.c":"3.4.5",
"ruuvi.libraries.c":"3.0.0",
...
"radio":{
"radio":{ "init":"pass,"
"radio":{ "init":"pass," "address_set_get":"pass"
"radio":{ "init":"pass," "address_set_get":"pass" },
"ble_adv_1_mbit":{ "init":"pass", "interval":"pass", "extended":"pass", "power":"pass", "scan":"pass" },
"ble_adv_2_mbit":{ "init":"pass", "interval":"pass", "extended":"pass", "power":"pass", "scan":"pass" },
"ble_gatt_1_mbit":{ "init":"pass", "tx":"timeout" "throughput":"0.000 bps" }, 
"ble_gatt_2_mbit":{ "init":"pass", "tx":"timeout" "throughput":"0.000 bps" },
"nfc":{ "init":"pass", "tx_rx":"timeout" }, }
}app_heartbeat.c:116 WARNING: NO_MEM
Movement

DG12, 2021-01-07:
Some tests should report FAILED if time out or throughput < 98% of expected. 
"ble_gatt_1_mbit":{ "init":"pass", "tx":"timeout" "throughput":"0.000 bps" },
"ble_gatt_2_mbit":{ "init":"pass", "tx":"timeout" "throughput":"0.000 bps" },
"nfc":{ "init":"pass", "tx_rx":"timeout" }, }

ojousima, 2021-01-08:
> PLEASE reduce the ble_gatt_1_mbit Tx test time. RuuviTag got hung and I waited a long time to complete test!!

Timeout takes time to let user interact, I agree that it's somewhat long if one wants just to run the program in debug mode. It would be possible to setup another variant for running tests, but the use case affects so few people that having extra variants which aren't needed 99.99% of time isn't sensible

> Actually is calculation correct?? 1_mbit tx @ 6.876kbps ? 2_mbit is not twice as fast??

Actually, it should be kBps, as in bytes, not bits. 2 Mbit/s modulation isn't necessarily twice as fast, it depends on the radio link quality. In noisy environments, 1 Mbit/s can actually be faster, but 2 Mbit/s is still more energy efficient because nRF52 can sleep while waiting for the next connection interval to retry sending data which was lost.

DG12, 2021-01-08:
How / way can the user interact with during the ble_gatt_*_mbit test?

I thought the the tests only did transmits. What causes the variation from run to run.

ojousima, 2021-01-08:
> I thought the the tests only did transmits. What causes the variation from run to run. 

GATT is acknowledged at link layer, so if acknowledgements are missed by the receiver side of test the data rate is slower. Fastest data rates I've seen can be achieved with a nRF52 DK running the system test binary.

closed 2021-01-25 by DG12