ruuvi/ruuvi.firmware.c #184: 3.29.3 NFC "Data" string is not properly initalized- defect; priority:L

DG12, 2020-12-18:
**in ruuvi.drivers.c/src/tasks/ruuvi_task_nfc.c**
**rt_nfc_init** has 2 arguments:  **(ri_comm_dis_init_t, * const init_data)**

 **ri_comm_dis_init_t**   is not used.

version 2.5.9 had   argument of data and data size (unfortunately that code is also incorrect)

Perhaps a common function would be better passing it 3 arguments :the label, the associated indicator(?) and the data.

I would be willing to fix this and  generate a pull request but since my 
**This branch is 1 commit ahead, 2 commits behind ruuvi:master.**
How do I synchronize my fork?  
In the current case there is only a slight spelling correction I made to read.md but what if there was 2 or 3 more significant differences.??

ojousima, 2020-12-18:
That comma looks like some kind of an error in your environment, neither https://github.com/ruuvi/ruuvi.drivers.c/blob/a9e9475a08a4705a8d70b1f7710f85a2e105efbb/src/tasks/ruuvi_task_nfc.h#L52 or https://github.com/ruuvi/ruuvi.drivers.c/blob/a9e9475a08a4705a8d70b1f7710f85a2e105efbb/src/tasks/ruuvi_task_nfc.c#L194 has it. 

Synchronizing the fork depends on changes, you can probably run either git merge or git rebase.

ojousima, 2020-12-18:
The data field has RAWv2 data being broadcasted, which isn't parseable as UTF-8.

closed 2021-03-19 by ojousima