ruuvi/ruuvi.firmware.c #186: 3.29.3 LED should toggle during initial HW test  - enhancement: priority:L

DG12, 2020-12-18:
It would be helpful if each call to a hardware test toggled an LED.
Currently the RuuviTag appears dead and reset doesn't seem to help( since it restarts the testing)
This is minor if it is a development user with a monitor connected but can be scary if a user inadvertently  loads a  test version (or as in my case I started a new day by plugging in the devboard end forgot to start up the monitor and I panicked because I thought the RuuviTag was bricked :-) )

ojousima, 2020-12-19:
For user who inadvertently loads a debug firmware, the end result would look like "tag just blinks and does nothing", so it really would not solve the issue. Overall, this shouldn't affect anyone who doesn't come to GitHub to try out new things, and users here are knowledgeable enough to debug their issues. Marking as won't fix.

DG12, 2020-12-19:
From my own experience seems worth the small amount of effort.

closed 2021-01-25 by DG12