ruuvi/ruuvi.firmware.c #187: 3.29.3 log entries occur more frequently than expected, may be related to sync - minor defect

DG12, 2021-01-01:
Here are timestamps from export after clearing and several sync attempts.
Those marked with A are from advertisements.
19:54:26
19:54:27
19:54:28
19:57:18A
19:59:28
19:59:29
20:04:28
20:04:29
20:08:51A
20:09:29
20:09:30

ojousima, 2021-01-04:
Logs are timestamped by Ruuvi Station, so getting duplicate entries is probably related to that. Deduplication would be good on app side, but firmware cannot know what logs the receiver already has.

DG12, 2021-01-04:
Although timestamps are added by Ruuvi Station, the ruuvi tag is transmitting more entries than the well defined frequency. 
Should I include more columns from the export file in the sample to clarify.

ojousima, 2021-01-04:
> Although timestamps are added by Ruuvi Station, the ruuvi tag is transmitting more entries than the well defined frequency.

1.

>> 19:54:26
19:54:27
19:54:28

2.

>> 19:59:28
19:59:29

3.

>>20:04:28
20:04:29

These look like they're duplicate entries, spaced by one second. It could be caused by difference in relative time to logs. Or do you mean something else?

DG12, 2021-01-04:
Thank you. I can't find this data so I will create a new set. It's hard to tell if they are duplicates as the sensor readings  are likely to be the same for samples at that small an interval .

closed 2021-03-19 by ojousima