ruuvi/ruuvi.firmware.c #188: 3.29.3 Sync fails intermittently - defect

DG12, 2021-01-01:
Flashed 3.29.3+default this morning
Attempt to sync to station IOS this evening from .1 meter away, fails "failed to serve in 60 seconds Serve aborted 

Second attempt same results.

Third attempt (I'm stubborn ) successful, 134 data points ate 5 minute intervals from ruuvi log as well as 6 collected by station IOS from advertisements while in range.

Clear.

10 +/- minutes later Sync , export OK.
5 +/- minutes later Sync, export OK.
5 +/- minutes later Failed!
5+/- minutes later Sync, export OK 20 data points (including 4 from advertisements).

Next , Android station exhibits similar problem!
Connecting ...Disconnected.
connecting.. wait for it ... Disconnected.
Connecting ...Disconnected.
Connecting ...Disconnected.
Connecting ...Disconnected.
connecting.. wait for it ... Disconnected.
Connecting ...Disconnected.  I'm not that stubborn , I give up :-(

ojousima, 2021-01-04:
There is a timeout after 5 minutes, maybe the firmware will get stuck into some invalid state after a timeout. Thanks for the report, we'll try to replicate.

@markoaamunkajo do you have time to check this in near future?

markoaamunkajo, 2021-01-04:
I will add card to check firmware, thanks!

DG12, 2021-01-04:
The time between start sync and abort is much less than 5 minutes.
The 5 minutes mentioned in the original issue is the time between my attempting to sync.

ojousima, 2021-01-04:
Is this when syncing with iOS version of Ruuvi Station? I think the iOS Station has 60 second timeout. It used to be followed by a watchdog reset by tag at 120 seconds, but #164 and #175 were fixed by kicking the GATT by tag at 300 seconds and triggering watchdog at 360 seconds.

DG12, 2021-01-04:
I had similar problems with both IOS and android versions. I don't think it is wdog related. Can you make up a firmware I can run on DevBoard  with "default" timings ,  NRF_LOG enabled and something like:

```
void app_error_fault_handler(uint32_t id, uint32_t pc, uint32_t info)
{
    watchdog_feed(); // dont want to fault
   
    NRF_LOG_ERROR("\r\n\r\n    Fatal. ID:%08x, PC:%08x, INFO:%08x, +++++++++++\r\n\r\n",id,pc,info);
    NRF_LOG_FINAL_FLUSH();

    for ( int16_t i=0; i<13; i++) { RED_LED_ON; nrf_delay_ms(500u); RED_LED_OFF; nrf_delay_ms(500u); }

 NVIC_SystemReset();
```

ojousima, 2021-01-04:
The current debug firmware already prints app errors to RTT console, https://github.com/ruuvi/ruuvi.firmware.c/releases/download/v3.29.3/ruuvitag_b_armgcc_ruuvifw_test_v3.29.3_full.hex . Please note the initial extended self-test takes a minute or so.

DG12, 2021-01-04:
I was concerned that  the high frequency of advertisements in debugging will affect to sync. I will try it out and post more details.

ojousima, 2021-01-04:
Advertisements and measurements are stopped while sync is in progress so that should not be an issue.

closed 2021-03-19 by ojousima