ruuvi/ruuvi.firmware.c #189: 3.29.3 under j-link : halt; sleep 1ms; go; sleep 2ms; halt causes HardFaultMemManage

DG12, 2021-01-04:
Is this simply an anomaly while running under the debugger??

I would not think that a 1 and 2 millisecond sleep should cause the wdog to time out (If that is what happened??)

See [MyBeacons/issues/HardFaultMemManag](http://mybeacons.info/issues/HardFaultMemManag.html) for details including stack dump.

DG12, 2021-01-05:
I am confused about the WDOG configuration discussed at 40.4.7 CONFIG
 of nRF52832 which says that if bit 3 (0x08) is reset "Pause watchdog while the CPU is halted by the debugger" .
I added mem32 4001050C,1  after the sleep 1 and it shows 4001050C = 00000001 .

My experience has been that under j-link issuing a halt , then inspecting a few memory locations , then a go causes a fault.
 Can you clarify that for me?

ojousima, 2021-01-05:
Halting program while Softdevice is running causes the Softdevice to miss Bluetooth timing constraints which leads to a hard fault. It certainly does make debugging BLE a lot harder.

DG12, 2021-01-05:
Thanks for that info.

closed 2021-01-05 by DG12