ruuvi/ruuvi.firmware.c #190: 3.29.3 Need mechanism to clear log - deficiency

DG12, 2021-01-07:
There needs to be a command from RuuviStation to clear the logged data in the RuuviTag.
See

ojousima, 2021-01-08:
Why this mechanism would be needed?

DG12, 2021-01-08:
For Example:
Reset does not clear log. Erase and flash firmware clearing out log.
Take the Ruuvi to the swamp, on Monday and leave it there.
Return on Wednesday and sync with station leave Ruuvi there. Get home export data to laptop.
Return on Friday sync gets data since Monday! 

Desired action is after sync on Wednesday, issue log reset so in Friday only get data not previously retrieved.
++++++++++++++++++++++++++++++++++++
Wait! This is a deficiency in Station only!  Log data in RuuviTag should not be cleared.

Second user arrives for the first time on Thursday. They should be able to get data since installation on Monday.

closed 2021-01-08 by DG12

ojousima, 2021-01-09:
Ruuvi Station already requests only for data which hasn't been fetched earlier, in your case Station would ask "All data" on Wednesday and "Data since Wednesday" for you, and "All data" on Thursday for your friend.