ruuvi/ruuvi.firmware.c #199: 3.30.0: Test binaries

ojousima, 2021-01-25:
Test process is described at https://docs.ruuvi.com/ruuvi-firmware/3.x/releases, 

The 3.x and 3.29 checklists need to be compared to one another, and any updates from 3.29 versions need to be moved under 3.x checklist. 

Firmware needs to be checked on as many board variants as possible.

niksiboxi, 2021-03-24:
Tested with RuuviTag B+ & RuuviTag B+ (SHTCX)

niksiboxi, 2021-03-24:
@ojousima Will you test the RuuviTag Basic?

ojousima, 2021-03-24:
@niksiboxi yes, we'll also need to redo the power measurements on RC7

niksiboxi, 2021-03-25:
@ojousima  Ok, will do the tests asap.

niksiboxi, 2021-03-25:
Added 3.30-RC7 power measurements.

niksiboxi, 2021-03-29:
@ojousima Is there any tests to be done still missing?

closed 2021-04-01 by niksiboxi

DG12, 2021-04-01:
Is this expected?

"nRF5TMP": {
"init":"pass",
"modes":"pass",
"configuration":"**fail"**,
"interrupts":"skipped",
"data":{
"timestamp_ms": "ld",
"temperature_c": "22.75"
}

ojousima, 2021-04-02:
>Is this expected?
>
>"nRF5TMP": {
>"init":"pass",
>"modes":"pass",
>"configuration":"fail",
>"interrupts":"skipped",
>"data":{
>"timestamp_ms": "ld",
>"temperature_c": "22.75"
>}

@niksiboxi is working on it