ruuvi/ruuvi.firmware.c #207: package.sh  ruuvitagb  nrfutil has unknown  option --sd-id  - 3.30.0 - defect

DG12, 2021-01-27:
package.sh  ruuvitagb (B0)  nrfutil has unknown  option --sd-id  - 3.30.0 - defect

package.sh for targets kaarle (CA) and 
 keijo and kalervo seem incomplete

ojousima, 2021-01-28:
That's related to SDK DFU update, which is relevant only for RuuviTag B.

DG12, 2021-01-28:
I didn't mean that --sd-id in unknown to ME, rather nrfutil version 2.3.0 fails.
Am I using wrong version of nrfutil?

ojousima, 2021-01-28:
A link to the exact line(s) where the problem is?

DG12, 2021-01-28:
Sorry.
line 38
nrfutil pkg generate --application _build/nrf52832_xxaa.hex --application-version 1 --hw-version 0xB0 --sd-req 0xB7 --key-file ruuvi_open_private.pem ${BINNAME}\_dfu_app.zip

DG12, 2021-01-28:
Sorry line 39
nrfutil pkg generate --application _build/nrf52832_xxaa.hex --application-version 1 --bootloader ${BOOTLOADER} --bootloader-version 2 --hw- +version 0xB0 --sd-req 0x91,0xAF,0xB7 **--sd-id 0xB7** --softdevice ../../../../nRF5_SDK_15.3.0_59ac345/components/softdevice/s132/hex/         +s132_nrf52_6.1.1_softdevice.hex  --key-file ruuvi_open_private.pem ${BINNAME}\_sdk12.3_to_15.3_dfu.zip --debug-mode

ojousima, 2021-01-28:
I'm guessing that --sd-id is newer than your nrfutil, my version is:
```
Otso-MacBook-Pro:ruuvi.drivers.c otsojousimaa$ nrfutil version
nrfutil version 5.2.0
```

Please update your nrfutil and try again, let's reopen this if update doesn't solve the problem.

closed 2021-01-28 by ojousima

DG12, 2021-01-28:
GAWK! How is my version so much older(?) (rhetorical question).

GAWK GAWK!! 
 pip install nrfutil
Collecting nrfutil
  Downloading nrfutil-6.1.0.tar.gz (842 kB)
     |████████████████████████████████| 842 kB 4.8 MB/s 
     and many others!!
now
ls -l `which nrfutil`
-rwxr-xr-x  1 dgerman  staff  235 Jan 28 08:16 /usr/local/bin/nrfutil*

nrfutil version
Traceback (most recent call last):
  File "/usr/local/bin/nrfutil", line 5, in <module>
    from nordicsemi.__main__ import cli
  File "/usr/local/lib/python3.7/site-packages/nordicsemi/__main__.py", line 52, in <module>
...
   from serial.tools.list_ports_osx import comports
  File "/usr/local/lib/python3.7/site-packages/serial/tools/list_ports_osx.py", line 32, in <module>
    kIOMasterPortDefault = ctypes.c_void_p.in_dll(iokit, "kIOMasterPortDefault")
ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found

DG12, 2021-01-31:
Apparently Big Sur no longer defines kIOMasterPortDefault.
Use older older version until Nordic update.