ruuvi/ruuvi.firmware.c #210: Transmitting log(history) in advertisement format - enha

DG12, 2021-02-03:


DG12, 2021-02-03:
I realize that the long term plan is to support many different sensors (some even for existing measurements)
It seems that since current systems to receive data (particularly ruuvistation and ruuvistation IOS) already have code to process the format rawV2(05) it would be worth the level of effort to transmit the log in advertisement format. 
Along with actual storing the data in the ruuvitag in advertise format would (see issue #99) would actually reduce the code size in the ruuvitag.

I would be willing to contribute toward this effort.

ojousima, 2021-02-04:
There certainly would be benefits in writing code for the one existing special case, and if there is some use case where extra space in flash is critical we might make a branch which has the more compact form. However, for all existing gateway implementations the current memory range is enough, so adding complexity by adding this special case in the mainline firmware is not beneficial.

closed 2021-03-19 by ojousima