ruuvi/ruuvi.firmware.c #211: Host based system to download log(history) and insert to database - proposal

DG12, 2021-02-03:
Is there a project currently to in progress to that would be able to retrieve the log(history) data and insert it directly into a databa
se.

Something using a similar scheme as the (really great) RuuviCollector.

This would allow deploying several (even many) ruuviTags. 
Retrieving them and simply placing them within range of the host for a few seconds to transfer the data.
Then redeploying them again.

The only accommodation to the ruuvtag might be a confirmation flash that the transfer is complete.

ojousima, 2021-02-04:
Yes, but the project is not public at least yet. I pinged the authors about this.

bchwtz, 2021-02-04:
We are currently working on a solution for this and have some prototype. Key for us is a solution that works in Python on a Raspberry Pi. Challenges are the deprecated gatttool from BlueZ and the unsure future of the python dbus implementations for addressing bluetooth devices apart from some architectural decisions. Our approach is far from a finished piece of software. We are however interested in discussions and open for collaborators. Fell free to reach out to me here or via Slack.

closed 2021-03-26 by ojousima