ruuvi/ruuvi.firmware.c #221: Integration test : sensors incorrectly includes 2 nulls and skips BME280 - 3.30.0 rc5 - defect

DG12, 2021-02-28:
Running ruuvitag_b_armgcc_ruuvifw_test_v3.30.0-RC5_full.hex from releases
(Slightly reformatted)
2  occurrence of
` ruuvi_nrf5_sdk15_log.c:68 WARNING: NULL `
occur amidst the null ( " ") sensor names

 ```
 "sensors": {
    " ": { "init":"skip" },
    " ": { "init":"skip" },
    "BME280":  { "init":"skip" },
    "nRF5TMP": { "init":"pass",
                 "modes":"pass",
                 "configuration":"fail",
                 "interrupts":"skipped",
                 "data":{ "timestamp_ms": "ld", "temperature_c": "25.00" }
                },
    "LIS2DH12": { "init":"pass",
                  "modes":"pass",
                  "configuration":"pass",
...
```

 On the same ruuviTag (6E29) rc2 reports:
```
...
"sensors": {
      "": {"init":"skip",},
      "": {"init":"skip",},
     "BME280": {init":"pass",
                "modes":"pass",
                "configuration":"pass",
                "interrupts":"skipped"
                "data":{
                   "timestamp_ms": "ld",
                   "humidity_rh": "25.88",
                   "pressure_pa": "100539.37",
                   "temperature_c": "23.36",
                      }
                },
    "LIS2DH12": { "init":"pass",
                  "modes":"pass",
                  "configuration":"pass",
..

```

ojousima, 2021-03-01:
Yes, the skipped sensors should be printed in a prettier way

DG12, 2021-03-01:
What are the 2 sensors that produce the nulls, and why is BME280 skipped ?

ojousima, 2021-03-02:
BME280 is not skipped on your board, data is 
```
"data":{
                   "timestamp_ms": "ld",
                   "humidity_rh": "25.88",
                   "pressure_pa": "100539.37",
                   "temperature_c": "23.36",
                      }
```
Skipped sensors are SHTC3 and DPS310, as you don't have them on your board.

ojousima, 2021-03-02:
The code also needed style checker run, continuing at https://github.com/ruuvi/ruuvi.firmware.c/pull/223 . The version history will retain attribution to your commits, closing this one. 

You'll have a lot easier time with your repository if you develop in a separate branch from master. If you develop in your master and it gets merged to ruuvi/master, after merge your and Ruuvi's master branches are out of sync and you'll need to delete your own master and replace it with Ruuvi's

If you develop in a separate branch, you can just pull updates from ruuvi/master once your work is merged.

closed 2021-03-02 by ojousima

DG12, 2021-03-02:
Regarding the skipped BME280.
notice that the FIRST set of data , from rc5 , shows   "BME280":  { "init":"skip" },

The seconds set of data is from rc2 which shows \..."humidity_rh": "25.88",...

DG12, 2021-03-02:
The first set is from RC5 !

ojousima, 2021-03-02:
Ok, then there must be something strange between RC2->RC5. I did make some adjustments in BME driver, maybe it is related.

DG12, 2021-03-02:
Regarding comments: Those  in version history in git is on a pull basis, that is really not helpful on a particular source file.
I have found, working with various version control and project source management tools, that changes documented in the source, which are obvious when reviewing the source, are much more helpful.

Consider the review you are about to make. regarding BME280. Are you going to review the git comments to see if and what changes were made?

ojousima, 2021-03-02:
Yes, relevant change sit is in this commit: https://github.com/ruuvi/ruuvi.drivers.c/commit/46a8c6bbef9525756bce90604671f493379cf4cd#diff-9cb970bfe83594d069f68239b5d39ce70e51bee32348adafa8f05d4aa3188fef

DG12, 2021-03-02:
Consider also version control history, even when detailed enough, are nearly impossible to resolve and understand in cases like :
1) add a check against maximum.
2) add a check against minimum, should have added that in update #1
3) revise bounds checks against better spec.
4) also check for zero.
Even if you review the version control comments in this trivial scenario  it is very hard to follow. Partially because the comments cannot be changed
Instead a comment in the source need not show the actual transactional history but more importantly what the current state is for example
1) Check against lower & upper bounds and zero too