ruuvi/ruuvi.firmware.c #229: 3.29.3: repeats advertisements really fast in 90 second cycle in some case

ojousima, 2021-03-12:


ojousima, 2021-03-12:
[image]
Not exactly 90 sec. 19:11:27 , 19:13:03, 19:14:40, 19:16:16

andreevdenis, 2021-03-12:
Duration is about 5 seconds. Rebooting sensor didn't help

DG12, 2021-03-12:
@ojousima What did you use to generate that display?

ojousima, 2021-03-12:
@ojousima What did you use to generate that display?

@andreevdenis console logs

DG12, 2021-03-12:
> What did you use to generate that display?
> 
> @andreevdenis console logs
Can you please provide a little more detail.

andreevdenis, 2021-03-12:
@DG12 This screen is from Android Studio. I was doing https://github.com/ruuvi/com.ruuvi.station/issues/166 and put logging for cases when BT library should filter measurement from sensor. One of my sensors was acting wierd

ojousima, 2021-03-19:
@niksiboxi please check if this can be repeated on 3.30 somehow

niksiboxi, 2021-03-26:
@ojousima Could not reproduce.

Tested with GATT -> Read attributes/subscribed to NUS notifications.

ojousima, 2021-03-26:
@niksiboxi This would be in advertisements, not in NUS notifications

niksiboxi, 2021-03-29:
@ojousima Could not reproduce -> Tested advertisement.

closed 2021-04-02 by ojousima