ruuvi/ruuvi.firmware.c #242: Flash must not be purged on boot, only on explicit request - defect

DG12, 2021-04-07:
System can boot for various reasons and flash, including configuration and logging history, must not be purged unless explicitly requested by user. 
This request might be via GATT or NFC with appropriate authentication or by button hold.

Consider the following FREQUENT occurrence: RuuviTag is deployed outside.  Various sensor readings are saved to log history. Temperature changes (either significant increase or decrease) causing battery voltage to drop below Brownout reset value and eventual system shutdown. Subsequent restoring of temperature to a level where battery now provides sufficient voltage to trigger power-on-reset. This scenario  must NOT clear configuration or history log. This scenario has been observed multiple times during a single battery lifetime.

Another senecio is one where the battery is depleted. Installing a new battery or otherwise providing power must NOT clear configuration or reset the history log.

ojousima, 2021-04-08:
Currently only logged data is erased on boot by default. This is because there is no way for the tag to know how long power outage was so timestamps of data would be unreliable. Bad data is often worse than no data.

closed 2021-04-08 by ojousima

DG12, 2021-04-08:
I think you should let the user decide. 
Although timestamps may not be valid, much important information is still available.
For example temperature that was at least as high/low as xxx. Even it xxx is not the highest because of highest being when tag was "offline" "at least as" may be very important.  In the "cold chain tracking" environment determining that temperature exceeded (or fell below ) xxx for at least mmm minutes may be more important than when it occurred.
If the user had recorded when the new battery was inserted or when a manually initiated reset was done they can know what the correction for the first period is.

Approximate time may be all the user needs and that may be able to be corrected for. An example could be multiple tags nearby measuring temperature.  

Again letting the user decide may be the best option.
Please reconsider.