ruuvi/ruuvi.firmware.c #257: 3.30.2-debug: ARMGCC version prints partial test logs, doesn't print U64 timestamps.

ojousima, 2021-05-27:


DG12, 2021-05-28:
Are you referring. to 
**"timestamp_ms": "ld",**

verses
"timestamp_ms": "41381",
when run via Segger Embedded Studio

ojousima, 2021-05-28:
> Are you referring. to
> "timestamp_ms": "ld",

yes

DG12, 2021-05-28:
It would seem the the ARMGCC toolchain version of printf does not handle %l  (ell)  ?

The difference is very nasty! Attempting to debug the ARMGCC version when SES uses different library.

Makes one wonder what other FUNCTIONAL differences there might be?

ojousima, 2021-05-28:
> Makes one wonder what other FUNCTIONAL differences there might be?

Yes it does. The integration and system tests are done on ARMGCC binaries though

ojousima, 2021-07-26:
Fixed in https://github.com/ruuvi/ruuvi.drivers.c/pull/221, thanks @DG12

closed 2021-07-26 by ojousima