ruuvi/ruuvi.firmware.c #266: v3.30.3+RC1: TMP117 consumes excessive power

ojousima, 2021-07-16:
3.30.2 consumes 42 uA on B8+TMP117:
[image]

3.30.3+RC1 consumes 52 uA on B8+TMP117:
[image]

3.30.2 consumes 37 uA on B8
[image]

3.30.3+RC1 consumes 34 uA on B8
[image]

It seems that TMP117 is not properly configured to not oversample in 3.30.3

ojousima, 2021-07-26:
Found the reason, boards I have have too weak pull-ups to configure TMP at 400 kHz. Retesting with stronger resistors. Related to #262

DG12, 2021-07-27:
Can you add which resistors were changed and from which value to which better value?

ojousima, 2021-07-27:
> Can you add which resistors were changed and from which value to which better value?

RuuviTag B8 [published schematic](https://github.com/ruuvi/ruuvitag_hw/blob/master/ruuvitag_revb8/ruuvitag_revb8_schematic.pdf) has correct 1k value, our initial try was 4k7. No need to write old values down, as they weren't published just in case we'd need to tune them.

DG12, 2021-07-27:
Surprised to see reset button removed.

ojousima, 2021-07-30:
Fixed in #271

closed 2021-07-30 by ojousima