ruuvi/ruuvi.firmware.c #281: nRF reboots if trying to pair using Android's Bluetooth manager

laurijamsa, 2021-08-21:
If I try to pair with Android, RuuviTag reboots, measurement sequence number resets and history gets lost.

[image]

Tried with Samsung Galaxy S21 Ultra and Lenovo tablet.

Defective firmware versions: 3.30.2 and 3.30.3 (and possibly older ones).

ojousima, 2021-09-08:
Possibly fixed in #285, bonding was disabled in config. Needs verification

niksiboxi, 2021-11-18:
The RuuviTag crashes with error log 
`ruuvi_nrf5_sdk15_communication_ble_gatt.c:365 FATAL: INVALID_STATE`

niksiboxi, 2021-11-29:
Possible solution is to handle the Android pairing process with the peer manager, hence the `BLE_GAP_EVT_SEC_PARAMS_REQUEST` is not needed.

ojousima, 2021-12-28:
@markoaamunkajo please check if this is fixed in 3.31.2-RC1

markoaamunkajo, 2022-01-10:
@ojousima Still reboots tag rev B7 and  B8.

ojousima, 2022-01-12:
@niksiboxi Can you take a look?

laurijamsa, 2025-03-27:
Possibly fixed a long time ago by removing a pairing flag. Needs to be double checked.

closed 2025-05-13 by ojousima