ruuvi/ruuvi.firmware.c #305: app_log anomalies: test record remains

DG12, 2021-11-28:
After some extensive investigation with the release version of the firmware.
I have found several anomalies for which I will create separate issues.
After running  for several hours
```
"firmware":"Ruuvi FW  v3.31.1",
"compiled_on":"Nov  8 2021 18:54:11",
"board":"RuuviTag B",
"ruuvi.boards.c":"3.8.0",
"ruuvi.drivers.c":"3.9.2",
"ruuvi.libraries.c":"3.0.0",

NFC SW: Ruuvi FW v3.31.1+debug
```

I have used j-link with he following to investigate these using:

view headers for FDS records "live"
(Copy these and paste directly into j-link)

```
mem32 00065000 10    // flash page  0 (16 pages)
mem32 00066000 10    // flash page  1
mem32 00067000 10    // flash page  2
mem32 00068000 10    // flash page  3
mem32 00069000 10    // flash page  4
mem32 0006A000 10    // flash page  5
mem32 0006B000 10    // flash page  6
mem32 0006C000 10    // flash page  7
mem32 0006D000 10    // flash page  8
mem32 0006E000 10    // flash page  9
mem32 0006F000 10    // flash page  A
mem32 00070000 10    // flash page  B
mem32 00071000 10    // flash page  C
mem32 00072000 10    // flash page  D
mem32 00073000 10    // flash page  E
mem32 00074000 10    // flash page  F

```
save FDS and view it with an editor or grep it

```
J-LINK> savebin 2111081854.bin 60000 15000

> hexdump 2111081854.bin >2111081854.hex # convert to text

```
+++++++++++++++++++++
 this page has a fileID of 2 and a record ID of 1
ruuvi.drivers.c/src/integration_tests/ruuvi_interface_flash_test.c
```
#define F_TEST_PAGE                 0002    // i.e. fileID
#define F_TEST_RECORD           0001

```
```
J-Link>mem32 00070000 10    // flash page  B this page has a fileID of 2 and a record ID of 1
00070000 = DEADC0DE F11E01FE 00050001 00000002 
00070010 = 000016DE 73616C46 65742068 64207473 
00070020 = 20617461 00000031 03EDF009 000000F0 
00070030 = 0000175A 0000D104 0000D2BF 000000FA 

J-Link>mem32 00071000 10    // flash page  C normal data page fileID of F0 record ID of 0A
00071000 = DEADC0DE F11E01FE 03EDF00A 000000F0 
00071010 = 0000175B 0000D2BF 0000D479 000000FA 
00071020 = 00000000 00000000 0000D2C3 41E8020C 
00071030 = 41B8F7CF 47C27B22 0000D2C4 41E8126F 

```

ojousima, 2021-11-29:
Thank you for the detailed investigation. There is some extra space for data other than logs, so having the test record is not critical but of course they should be cleaned up.

ojousima, 2021-11-29:
The other anomalies might be related to #282 , reopening that if necessary

closed 2021-12-28 by ojousima