ruuvi/ruuvi.firmware.c #307: app_log anomalies: pages 60000..63000 are not init by FDS

DG12, 2021-12-01:
Last flash page NOT belonging to FDS {GOOD}:
```
J-Link>mem32 5F000 10
0005F000 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
```
(Chip Erase did it)

FDS should have  60000..75000(actually 74FFF)

...
```
ruuvitag_b.ld:
  .storage_flash 0x60000 (NOLOAD) :
  {
    __start_storage_flash = .;
    KEEP(*(SORT(.storage_flash*)))
    __stop_storage_flash = . + 0x15000;
  } > FLASH
```

pages 0,1,2,3 missing FDS header {BAD} :
```
J-Link>mem32 00060000 4   // flash page  0 (20 pages)
00060000 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
J-Link>mem32 00061000 4   // flash page  1
00061000 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
J-Link>mem32 00062000 4   // flash page  2
00062000 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
J-Link>mem32 00063000 4   // flash page  3
00063000 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 

```
Lowest address belonging to FDS {GOOD}:
```
J-Link>mem32 00064000 0C   // flash page  4
00064000 = DEADC0DE F11E01FE FFFFFFFF FFFFFFFF 
```
...
test page remaining See issue #305 
```
J-Link>mem32 00072000 0C   // flash page 12
00072000 = DEADC0DE F11E01FE 00050001 00000002 
00072010 = 000007A9 73616C46 65742068 64207473 
00072020 = 20617461 00000031 FFFFFFFF FFFFFFFF
``` 

Page reserved for Garbage Collection:
```
J-Link>mem32 00073000 0C   // flash page 13
00073000 = DEADC0DE F11E01FF FFFFFFFF FFFFFFFF 
                        01FF
```
boot_count & Config records:
```
J-Link>mem32 00074000 0C   // flash page 14
00074000 = DEADC0DE F11E01FE 00010000 000000F0 
00074010 = 000007AA 00000000 000200C3 000000CE 
00074020 = 000007AB 00000000 0000F401 00020028 
```

This sample was taken before the RAM block was filled and sent to FDS

ojousima, 2021-12-02:
This is probably caused by configuration `APP_FLASH_PAGES 16U`. In addition to the application-reserved pages there's one page for FDS garbage collection swapping, so 17 pages in total. 0x75-0x60 = 21, when we subtract 17 pages 4 pages remain. FDS ha no reason to initialize the bottom pages 0,1,2,3

DG12, 2021-12-02:
Although the correct calculation is 0x74-0x60 =20 since page 74000 is the last usable page(75000 is the  beginning of the area NOT available for FDS). The GC page is included in the flash page count. 20-16 STILL leaves 4 pages

DG12, 2021-12-02:
Previous comment retracted. No coffee yet, not even breakfast yet

closed 2021-12-16 by ojousima