ruuvi/ruuvi.firmware.c #309: app_log anomalies: blocks leave 3 element slots unused

DG12, 2021-12-02:
Not a biggie but this is a result of
```
app_log.h
#define STORAGE_RECORD_HEADER_SIZE (96U)
```
Actually
```
fds_internal_defs.h
#define FDS_PAGE_TAG_SIZE       (2) // Page tag size, in 4-byte words.
#define FDS_HEADER_SIZE         (3) //  record header words
```
so
`#define STORAGE_RECORD_HEADER_SIZE     (FDS_PAGE_TAG_SIZE+FDS_HEADER_SIZE)  *4 
and

`
```
J-Link>mem32 72F90 20
00072F90 = 41D58F5C 47C41054 000001B4 41D5353F 
00072FA0 = 41D6C6A8 47C4106C 000001B6 41D524DD 
00072FB0 = 41D7F9DB 47C4107A 000001B7 41D5C49C 
00072FC0 = 41D8D0E5 47C41092 FFFFFFFF FFFFFFFF 
00072FD0 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
00072FE0 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
00072FF0 = FFFFFFFF FFFFFFFF FFFFFFFF FFFFFFFF 
00073000 = DEADC0DE F11E01FE 03EDF001 000000F0 
```
Beginning of first record
```
J-Link>mem32 00072000 2C
00072000 = DEADC0DE F11E01FE 03EDF000 000000F0 //  FDS_PAGE_TAG(0..1) FDS_HEADER(0..1)
00072010 = 000007B0 00000000 000001B6 000000FA  // FDS_HEADER(2) ,   start_time, end_time, num_samples, 
00072020 = 00000000 00000000 00000000 41CFE979  // block_configuration;
00072030 = 41D976C9 47C41A8C 00000001 41CF8106 
00072040 = 41D953F8 47C41A8B 00000003 41CF147B 
00072050 = 41D8F5C3 47C41A68 00000004 41CF872B 
00072060 = 41D8ED91 47C41A5D 00000006 41CF2B02 

```
See also issue #308

ojousima, 2021-12-02:
Yes, there is intentionally a bit margin for header.

closed 2021-12-02 by ojousima

DG12, 2021-12-02:
Making a constant with a define does not make it a muggle value

#define STORAGE_RECORD_HEADER_SIZE (96U)
What's 96?
Should it be (FDS_TAG_SIZE+FDS_HEADER_SIZE+XTRA)*4

ojousima, 2021-12-02:
> Making a constant with a define does not make it a muggle value

True that. It's about filling 4096 byte page to 4000 bytes and leaving the rest for a header. Maybe PHYSICAL_FLASH_PAGE_SIZE - DATA_STORED_IN_PAGE_SIZE or something like that would be better

DG12, 2021-12-15:
I have seen a case where the beginning of the flash page had boot_count record in the beginning of the page and the log history record key starting at page+6 leaving only 10 words unused in that page.

ojousima, 2021-12-16:
> I have seen a case where the beginning of the flash page had boot_count record in the beginning of the page and the log history record key starting at page+6 leaving only 10 words unused in that page.

That's intentional behavior from FDS. It should be noted though that the data logging would be better served by direct access to flash rather than through FDS, but the drivers do not support it as of right now. Related issue https://github.com/ruuvi/ruuvi.drivers.c/issues/108