ruuvi/ruuvi.firmware.c #312: flash routines use  boolean for "busy" flag should it be a counter - question

DG12, 2021-12-12:
Is it possible that multiple actions could be queued and the first one to complete calls event handler and incorrectly sets processing false even though there is still an outstanding action?

ruuvi.drivers.c/src/nrf5_sdk15_platform/flash/ruuvi_nrf5_sdk15_flash.c

delete: 
does a record find, sets processing  true, then does delete; then may set processing  false

set:
checks processing, does record find, sets processing true, then either does an update or write and sets processing false

get: 
checks processing, does record find, open, close

GC: 
checks processing, sets processing true, calls GC and exits.

fds_evt_handler 
sets  m_fds_processing  false when an event occurs.

ojousima, 2021-12-13:
That's possible, access to flash busy flag should be protected by an atomic mutex.

Counter would not really help here, because only one part of program is supposed to access flash storage at a time, case where `busy` is two a bug has already occurred.

closed 2021-12-13 by ojousima

DG12, 2021-12-13:
When you have a chance 
can you elaborate on "case where busy is two a bug has already occurred.'

Should this stay open?