ruuvi/ruuvi.firmware.c #326: Tag became "Not connectable" suddenly and rebooting the tag made it connectable again on iOS

priyonto, 2022-02-15:
The tag was working fine until suddenly it stops showing it's connectable on the iOS app. The tag is on the latest firmware and when it was in "not connectable" mode the firmware update page showed something like this. 
[image]

I rebooted the iPhone, that did not fix the issue. But, rebooting the tag made it connectable instantly.

ojousima, 2022-02-16:
I can reproduce this issue on one of my 3.31.1 tags, looking into how I can get a tag to lock up like this

ojousima, 2022-02-16:
I opened nRF Sniffer to scan BLE traffic and checked the sensor I have issue with. I forced a disconnect on sensor by doing NFC scan and observed that there is a device connecting to the sensor almost right away. 

Wireshark screen capture, you can see first device running a scan request and second connecting:
[image]

I have no idea which device is connected to my tag, it might belong to someone in my household or my neighbor. I'd say this is not a firmware bug, but more of an user experience issue. It might be helpful to add a "Connect on NFC" feature where after NFC scan Ruuvi Station tries to immediately connect.

priyonto, 2022-02-16:
Does it mean it connecting automatically to another device when NFC enabled device is around?

DG12, 2022-02-16:
Can you take the sensor "out of range" of the unknown connector and see what happens? basement?

ojousima, 2022-02-17:
> Does it mean it connecting automatically to another device when NFC enabled device is around?

The flow is: 
1) Tag is connected to unknown phone A
2) My phone B does NFC scan. Phone A disconnects
3) Phone A reconnects before I have a chance to connect with my phone B

 > Can you take the sensor "out of range" of the unknown connector and see what happens? basement?

Later this month, I'm currently quarantined.

priyonto, 2022-02-17:
> The flow is:
> 
> 1. Tag is connected to unknown phone A
> 2. My phone B does NFC scan. Phone A disconnects
> 3. Phone A reconnects before I have a chance to connect with my phone B

This is super weird. I am wondering what is causing this behavior. This is not an app issue either, is it?

ojousima, 2022-02-17:
> This is not an app issue either, is it?

Not really, some other device is "hijacking" the sensor connection. We could secure the connection by using out-of-band pairing, such as NFC but it has been a design decision to use open connections for simplicity

closed 2022-07-22 by ojousima