ruuvi/ruuvi.firmware.c #331: Building without RT_FLASH_ENABLED fails

Scrin, 2022-07-22:
I'm trying to create a "minimalistic" firmware that more closely matches the "original" behavior of ruuvitags, namely I don't want the tag to be connectible and thus I don't want the tag to unnecessarily store measurements either (both of these for battery saving purposes, as I have no need for these features).

From `app_log.c` I found that setting `RT_FLASH_ENABLED` to false will replace the functionality with dummy functions, however doing this also breaks the build by causing a bunch of undefined reference errors. Looking at `app_config.h` it looks like this should work, since `RT_FLASH_ENABLED` is conditionally enabled there based on the available flash space. Am I missing something?

ojousima, 2022-07-22:
Thanks for the report. 

Both Ruuvi Flash functions and Nordic GATT functions depend on Nordic Flash libraries being available. Disabling Ruuvi Flash functions results in Nordic Flash functions not getting compiled, hence undefined references. 

I'll fix this in #332 by enabling lower-level Flash functions if GATT is enabled. I also added configuration to long-life variant to disable both GATT and Flash memory, because connecting to longlife tag is pretty difficult as advertisements are rare. That should also mimic 2.X firmware broadcasting RAWv2 quite well.

ojousima, 2022-08-23:
Fixed in #332

closed 2022-08-23 by ojousima