ruuvi/ruuvi.firmware.c #334: Gatt sync stalls

markoaamunkajo, 2023-03-02:
Description:

Gatt sync stalls in both Android and iOS.
Reported by multiple users.

This is related to Bluetooth Connection Length Extension.
Tag sends data as much as it can until something breaks the connection in interval and then it has to wait for the next interval.

1250 would be a flash page boundary, It makes me think that flash itself might have questionable data. That should be cleared in sensor reboot though.

Original issue: https://github.com/ruuvi/com.ruuvi.station.ios/issues/1069

DG12, 2023-03-02:
Sensor reboot is not necessary.
Restarting download may complete or stall at a different spot.

DG12, 2023-03-02:
Placing a sensor in the refrigerator works well to duplicate this problem.

ojousima, 2023-03-03:
What would be the fix here? We can change the parameters of connection, but we cannot have one set of parameters that will be the best option in every case.

DG12, 2023-03-03:
It seems that the app should "understand" that connection has been disrupted and either restart from the beginning or pick up from where it left off. 

My observation is that simply restarting the download (without any action to the tag) will continue sometimes succeeding.
Please review the initial submission of issue #1069. This is from an old version.
I will duplicate thee test and report back later today.

DG12, 2023-03-16:
Stall at 554