ruuvi/ruuvi.firmware.c #355: GPIO is not uninitialised on failed accelerometer initialisation, leaving floating pins

ojousima, 2023-12-12:
Possible fixes:
 - Uninit GPIO on failed init
 - Configure pull-up / down on GPIO

DG12, 2023-12-13:
How did you discover this?

ojousima, 2023-12-13:
I took off accelerometer with hot air and ran power profile

DG12, 2023-12-13:
What prompted you to do THAT ?

should this issue be in [https://github.com/ruuvi/ruuvi.drivers.c](url)
?

ojousima, 2023-12-13:
> What prompted you to do THAT ?

We experiment on all kinds of things ;)

> should this issue be in [https://github.com/ruuvi/ruuvi.drivers.c](https://github.com/ruuvi/ruuvi.firmware.c/issues/url)

Not really. Drivers report back to application correctly that the accelerometer was not properly initialised, problem is in main.c/app_sensor.c and `app_sensor_acc_thr_set()`